riscv/riscv-j-extension #19: nit on wording: Replace canonicity with canonicality

deepak0414, 2022-12-02:
I am not sure if `canonicity` is actual word.
I believe `canonicality` what is meant here.

martinmaas, 2022-12-07:
Good point – both words seem to be in use, but "canonicity" seems to be much more common (e.g., Google has ~900k hits for "canonicity" but only ~11k hits for "canonicality"). Merriam Webster [1] only has canonicity, Wiktionary [2] has both. From what I can tell, they are synonyms. Since "canonicity" seems more widely used, I would lean towards this one, but do not have a strong opinion either way.

[1] https://www.merriam-webster.com/dictionary/canonicity
[2] https://en.wiktionary.org/wiki/canonicality

pdonahue-ventana, 2022-12-07:
Reaching for my bookshelf, I checked my 1929 Funk & Wagnalls dictionary, my 1985 American Heritage dictionary, and my 2001 Random House dictionary.  All three only have "canonicity" so I think that word is canonical (I couldn't help myself).  Perhaps this is a US vs. UK thing, but none of them had a "Chiefly Brit." entry for canonicality as they do for other British words/spellings.

deepak0414, 2022-12-09:
Interestingly risc-v privileged spec never mentions use of word 'canonical' for treatment of high bits.
Although unpriv spec uses the word mostly for topics related to `NaN`

I am fine either ways. It was a nit.
So feel free to close this one.

martinmaas, 2022-12-09:
Sounds good. I will close for now, but we can revisit if it comes up again in AR or public review.

closed 2022-12-09 by martinmaas