riscv/riscv-j-extension #22: Zjpm: Differences in ignore transformation and sign extension when paging is enabled v/s not enabled

deepak0414, 2022-12-03:
Ignore transformation is explained mainly from perspective when paging is enabled.

When paging is not enabled (or bare mode), ignore transformation still apply?
I believe sign extension exist to accommodate negative kernel addresses (which is function of paging)

In case of M mode or bare mode, there is no paging and thus a sign extended address may fault (may not be even required)
Is the behavior same (of sign extension) when paging doesn't exist ?

martinmaas, 2022-12-09:
Addressed in fd7220807e1405ba9b5de38d9d4ac26e73312128 (Sections 2.1 and 2.4)

martinmaas, 2023-09-23:
The new version of the spec further clarifies this – now, virtual addresses are using sign extension and physical addresses are using zero extension (Section 2.2). Section 2.3 has examples of both. Physically addressed environments therefore no longer have to reason about NVBITS and VBITS.

I will close this issue for now – please feel free to reopen if there are still concerns about the latest version.

closed 2023-09-23 by martinmaas