riscv/riscv-j-extension #24: Zjpm: Restrictions on N for higher privilege with respect to lower privilege

deepak0414, 2022-12-03:
Spec doesn't put restrictions on values of N that can be programmed by lesser privileged.

As an example if S mode has configured a value of N = 18 for itself then U mode shouldn't be able to program anything > 18 for user pointers.

This is quite obvious and no implementation should allow this to happen.
But calling it out in spec will be be good.

martinmaas, 2022-12-09:
Addressed in fd7220807e1405ba9b5de38d9d4ac26e73312128 (Section 2.5)

deepak0414, 2023-01-23:
Section 2.5 says

`For example, an operating system's ABI may mandate that (e.g.,) user mode should never mask more bits than kernel mode`

This means that in order to enforce this kernel mode would never allow user to change it's N.
If that's the case, I believe user mode's config (upm) will always be dictated by kernel, rendering upm not very useful for all practical purposes.

martinmaas, 2023-09-23:
Since the latest version of the spec no longer supports delegation and only supports N = 16 across all privilege modes, I think this issue should be addressed now.

closed 2023-09-23 by martinmaas