riscv/riscv-j-extension #26: Zjpm: Addresses reported in CSRs (stval, sepc) hold which addresses?

deepak0414, 2022-12-03:
Once PM is enabled, spec must define which values (pre-transformed address, address after masking out bits or ignore transformation address) go into CSRs (sepc, stval on PF) which report addresses to higher priv mode.

Future extensions (like MTE, etc) might fault on mismatches and depending on how fault definition is done, this information will be needed to resolve such faults.

martinmaas, 2022-12-09:
Addressed in fd7220807e1405ba9b5de38d9d4ac26e73312128 (Section 2.6)

martinmaas, 2023-08-20:
The latest version of the spec (75e6aa4) now explicitly defines that hardware writes to a CSR store the transformed address (e.g., in the epc and stval cases listed above). On the other hand, software writes store the original address.

closed 2023-08-20 by martinmaas