riscv/riscv-j-extension #27: Zjpm: nit - small note on IOMMU and memory accesses

deepak0414, 2022-12-03:
Although it is quite obvious that `zjpm` spec concerns only with CPU based accesses.
Putting a note that device side accesses for shared virtual memory will be addressed by non-ISA spec will be useful.

Use cases like opencl [1] heavily use shared virtual memory between device side and cpu side.

[1] https://www.intel.com/content/www/us/en/developer/articles/technical/opencl-20-shared-virtual-memory-overview.html

martinmaas, 2022-12-09:
Addressed in fd7220807e1405ba9b5de38d9d4ac26e73312128 (Section 2.6)

deepak0414, 2023-01-23:
Instead of saying this

`Pointer masking only applies to accesses generated by instructions on the CPU (including CPU
extensions such as an FPU). For example, it does not apply to accesses generated by the IOMMU or
devices`

It should be on the line

`Zjpm specification concerns with pointer masking applied to memory accesses generated by instructions executing on the hart. Pointer masking on memory accesses from devices may be covered by non-ISA specifications like IOMMU specification`

martinmaas, 2023-09-23:
Quickly checking in about this item whether you still have this concern about the latest version of the spec. The suggestion from the architecture review committee was to remove this paragraph altogether since it is implied by the fact that pointer masking is an ISA extension.

deepak0414, 2023-09-29:
Yeah it was a nit anyways.
That's fine by me if we close it.

closed 2023-09-29 by deepak0414