riscv/riscv-j-extension #39: Does the top N bits need to be incremented by 1 or be unchanged, if there is a carry from the low bits during virtual address calculation?

stephen-ji, 2023-04-23:
When supporting ZJPM, for the AUIPC instruction, does the top N bits of the write back value need to be incremented by 1 or be unchanged if there is a carry from the low bits during  virtual address calculation？
Similarly, for a branch instruction,  if there is a carry from the low bits during target PC calculation, should the high N bits of target PC be incremented by 1 or remain unchanged?

stephen-ji, 2023-04-26:
(1)
When supporting ZJPM, for the AUIPC instruction, does the top N bits of the write back value need to be incremented by 1 or be unchanged if there is a carry from the low bits during result calculation？
For example:
Assuming that the CPU is currently in SV48 mode, and N of PM is 16，and current PC is 0xabcd_ffff_ffff_f000. There is an AUIPC instruction at this address (auipc x15, 0x1000). What's the write back value for x15? 0xabcd_0000_0000_0000 or 0xabce_0000_0000_0000?

Similarly, for a branch instruction, if there is a carry from the low bits during target PC calculation, should the high N bits of target PC be incremented by 1 or remain unchanged?
For example:
Assuming that the CPU is currently in SV48 mode, and N of PM is 16，and current PC is 0xabcd_ffff_ffff_fff0. There is an unconditional jump instruction at this address (jal x0, 0x10). What's the target address of this jump instruction? 0xabcd_0000_0000_0000 or 0xabce_0000_0000_0000?

(2)
In the case of ld or st instructions, when address calculation results in carry into masked bits, should the carry be discarded or retained?
For an ld or st instruction with a zjpm extension, when an exception occurs such as a page fault or access fault, should the high bits of the reported mt value bethe virtual address with pm information or the virtual address with sign extension applied?
For instance, for an ld instruction with an address of 0xf000_0000_0000_0000 and the number of pm=4, which results in a real address of 0x0, if a page fault occurs at that address, should the reported mt value be 0x0 or 0xf000_0000_0000_0000?

(3)
In RISCV debug with zjpm extension, is the execute/load/store address used for address trigger match before or after pointer masking？
For example:
When SATP.mode=SV39, and privilege mode is u-mode, upm.uinst=1, upm.uenable=1, upm.ubits=16, the effective fetch address is 0xabcd_0000_1234_5678, which address should be used for trigger match judgement: 0xabcd_0000_1234_5678(before mask) or 0x0000_0000_1234_5678(after mask)

martinmaas, 2023-09-23:
Apologies for the long delay! All three points should now be addressed in the latest version of the spec:

(1) This no longer applies, since instruction pointer masking has been removed from the draft.

(2) The address calculation is performed before the pointer masking transformation, so the carry would be discarded. In terms of which address is reported for exceptions, page faults, etc. the spec is now explicit that the masked portion is dropped in this case.

(3) The spec is now explicit that address trigger match applies pointer masking to the memory access address that is being matched.

closed 2023-09-23 by martinmaas