riscv/riscv-j-extension #47: Pointer Masking Enable 2-bit CSR field: why?

ingallsj, 2023-09-26:
Why is this field two bits wide?  What future use do we have in mind for the Reserved values 01/10?  Is it for separate high/low controls, similar to ARM's TCR_ELx.TBI0/1?

https://github.com/riscv/riscv-j-extension/blob/ac6782d43813a8674a4704ff01ef00555dfe73d1/zjpm/instructions.adoc?plain=1#L10-L20

martinmaas, 2023-09-28:
It is likely that we will want to support different values than N=16 in the future. Reserving the extra bit creates forward-compatibility for this anticipated scenario.

ingallsj, 2023-09-29:
I'm guessing that future smaller values of the `PME` field will correspond to smaller values of N?

martinmaas, 2023-09-29:
Either specific, smaller values of N, or we may use one value to indicate that N is dynamic/defined elsewhere/...

closed 2023-09-29 by ingallsj