riscv/riscv-j-extension #49: Pointer Masking Instruction Fetches

ingallsj, 2023-09-26:
Not applying pointer masking to instruction fetches will thus require address sanitization arithmetic still before every `jalr` instruction.  Why not?  That's not as huge of an overhead as every explicit load/store, but the inconsistency still caught my attention.

https://github.com/riscv/riscv-j-extension/blob/ac6782d43813a8674a4704ff01ef00555dfe73d1/zjpm/background.adoc?plain=1#L52

Point of comparison: ARMv8 applies TCR_ELx.TBI/0/1 to both data and instruction accesses' virtual addresses; but not page table walks, which are (intermediate) physical addresses.  It wasn't until the Pointer Authentication feature extension in ARMv8.3 that ARM added the TCR_ELx.TBID/0/1 fields to apply only to data accesses.  Was there no use case for the years of devices shipped based on ARMv8.0 - v8.2?

ingallsj, 2023-09-26:
I see I've duplicated https://github.com/riscv/riscv-j-extension/issues/30.  It would be great if there were minutes from

> per request from the architecture review committee and discussions in the J Extension group meetings.

martinmaas, 2023-09-28:
Here is the email describing the decision by the architecture review committee:

https://lists.riscv.org/g/tech-chairs/message/1393

TL;DR is that the committee wants to see a specific use case of instruction pointer masking before it is included in the standard. The current spec leaves the option open to add this in a future extension.

ingallsj, 2023-09-29:
I can't read that message, because I'm not a member of the `tech-chairs` mailing list.

But, okay, I only see references in the wild to pointer masking for stack and heap allocation, not instructions.

closed 2023-09-29 by ingallsj

martinmaas, 2023-09-29:
My bad – the message was cross-posted, here is the version that is world-visible:

https://lists.riscv.org/g/tech-privileged/message/1306