riscv/riscv-j-extension #50: Sign extension or no sign extension when V=1 but vsatp.MODE = Bare

deepak0414, 2023-09-28:
Section 2.2 (Ignore Transformation) in current spec (0.7) says that 

```
The ignore transformation differs depending on whether it applies to a virtual or physical address.
For virtual addresses, it replaces the upper N bits with the sign extension of the N+1st bit.
```

Additionally it says

```
When applied to a physical address, the ignore transformation replaces the upper N bits with 0.
This includes both the case of running in M-mode and running in other privilege modes with Bare
address translation mode.
```

Software running in VS / VU mode with vsatp.MODE = Bare (but hgatp.MODE != Bare)
From PM perspective is it considered non-bare address translation mode or bare translation mode ?
In this situation, PM is applying to VS / VU mode software and from their perspective its Bare mode.
And all they care about is their own address space which is physical and thus sign extension won't apply.
Is my understanding correct? Probably having this clarification somewhere in spec would be useful.

Let me know if I missed something or overthinking this and it is spelled out somewhere in spec.

martinmaas, 2023-09-29:
I agree this case should be treated as physical address translation (zero extension). We can add a clarification to the spec and I will keep this issue open until then.

martinmaas, 2023-10-21:
Added a non-normative comment in 92a88a2.

closed 2023-10-21 by martinmaas