riscv/riscv-j-extension #53: Better name for "N"

pdonahue-ventana, 2023-10-02:
The pointer masking extension defines the number of masked bits as N.  In this extension and elsewhere we have parameters with descriptive names like VBITS, XLEN, ASIDLEN, IALIGN, SEW, etc.  I think that N should be renamed to something that is more descriptive like MBITS, MASKBITS, MLEN, MASKLEN, etc.

Other (better) names that people sent to me on or off the list: NONABITS (non-address bits), HWALEN (HWASan), PMBITS, PMLEN.

For what it's worth, I think that *LEN is more RISC-V-like than *BITS.  (XLEN, FLEN, VLEN, ELEN, ILEN, ASIDLEN, VMIDLEN, etc.)

SiFiveHolland, 2023-10-03:
I would also add "PMASK" as a suggested prefix, as "PM" is overloaded with other interpretations (e.g. power management).

As for "BITS" vs "LEN", I don't see a clear winner. "LEN" is more popular, but the most closely related parameters (VBITS, NVBITS) use "BITS".

martinmaas, 2023-10-21:
I picked PMLEN and updated the spec in 92a88a2. The reason I decided against *BITS is that we consistently use LEN for scalars and BITS for ranges. "The PMBITS+1st bit from the top" would therefore be contradictory. Feedback welcome!

SiFiveHolland, 2023-10-21:
That makes sense to me!

pdonahue-ventana, 2023-10-21:
Sounds good.

closed 2023-12-09 by martinmaas