riscv/riscv-j-extension #56: ISA Extensions section clarifications

JamesKenneyImperas, 2023-12-04:
Can you please clarify some of the information in _ISA Extensions_ section?

1. There is a statement _All of these fields are read-only 0 on RV32 systems_. I don't think the term _RV32 system_ is precise enough. What is the behavior on a core with variable XLEN?

2. Under the `Smnpm` section there is a note that _The type of address determines which type of pointer masking is applied_. Is this still relevant? I see there is a recent change that stated that all addresses are treated the same, but I haven't looked into the details of that yet.

3. Normally the visibility and behavior of fields in `henvcfg` is governed by the same fields in `menvcfg`. Is that also the case here? If so, can that be stated explicitly? See the `Sstc` extension specification for an example of how this is described. If it isn't the case, it would help to have that explicitly stated too.

Thanks.

pdonahue-ventana, 2023-12-04:
> There is a statement All of these fields are read-only 0 on RV32 systems. I don't think the term RV32 system is precise enough. What is the behavior on a core with variable XLEN?

Also, it should be hart rather than system.  You can have heterogeneous systems.

martinmaas, 2023-12-09:
The term "RV32 system" follows the way this term is used in the relevant parts of the privileged RISC-V spec, describing the physical underlying register width before any XLEN reconfiguration. E.g., Section 3.1.6.2 of the privileged spec:

> For RV32 systems, the SXL and UXL fields do not exist, and SXLEN=32 and UXLEN=32.

The pointer masking standard (Section 2.7) describes the behavior in the presence of variable XLEN – if XLEN is set to 32 via UXL/SXL/MXL, the behavior is the same as on RV32.

I realize that "system" can have a range of meanings and the current definition should not preclude (e.g.,) SoCs that have different types of RISC-V processors in them. However, I think we should stay consistent with the definitions in the base ISA.

JamesKenneyImperas, 2023-12-09:
Thanks Martin, I think that makes sense.

I think a couple of minor additional clarifications might help. The _Pointer Masking Extensions_ section says:

> Note that in RV32, the CSR bits introduced by pointer masking are still present, for compatibility between RV32 and larger systems with UXL/SXL/MXL set to 1.

1. Does this mean that the PMM fields are visible with read-only non-zero values in the relevant `xenvcfgh` registers in RV32 mode?
2. Unless it's been recently changed, there isn't an `senvcfgh` register, so the statement above isn't true for the `senvcfg.PMM` field in the RV64 `senvcfg` CSR. Also, that CSR is shared between HS and VS modes, so the field will be cleared to zero on a transition from virtual mode with VSXLEN=32 to non-virtual mode with HSXLEN=64 (at least I think so - see issue https://github.com/riscv/riscv-isa-manual/issues/1173). This isn't necessarily a problem (the Hypervisor specification says that _Hypervisor software is expected to manually swap the contents of these registers as needed_) but I thought I'd point that out in case there is an expectation that it would be preserved in this scenario.

Thanks.

martinmaas, 2024-01-15:
I added clarifications to the corresponding sections stating that setting UXL/SXL/MXL to 1 clears the corresponding pointer masking bits (commit 8088461). I believe this should address both of these issues (if you have additional concerns, please re-open this issue).

closed 2024-01-15 by martinmaas