riscv/riscv-j-extension #58: Clarify why pointer masking is applied when xtval is written by hardware

JamesKenneyImperas, 2023-12-07:
The specification clearly states that pointer masking is applied to addresses before they are written to `xtval` registers by hardware. Could you please add a clarifying note about the reason for this? I can envisage some usage scenarios where the full value would be very useful in the trap handler, and a similar feature in another architecture I'm familiar with preserves the tag bits, so I'm curious why this is different.

Thanks.

martinmaas, 2023-12-09:
The rationale was discussed here: https://lists.riscv.org/g/tech-j-ext/topic/96872171#456

Delivering these bits may add overheads in some hardware implementations, and the pointer masking configuration is per privilege mode, so all trap handlers in supervisor mode would need to be careful to configure pointer masking the same way as user mode (which we don't mandate or assume) or manually unmask (which is expensive).

There was a sense in the discussion that there may be future use cases that require this, and future standards for these use cases may introduce a mechanism to deliver the mask bits to the exception handler.

Happy to add this rationale to the spec, but I wanted to first see whether this answers your question or whether you see current use cases that would run into problems with this behavior.

JamesKenneyImperas, 2023-12-09:
I've read that post and it makes sense now. I don't have any concrete existing use case in mind, so I'd be happy with just a clarification of the rationale in a note with the text you've given. Do you want to add that and close this issue?

Thanks.

martinmaas, 2024-01-15:
Done, added in 8088461.

closed 2024-01-15 by martinmaas