riscv/riscv-j-extension #75: Pointer Masking PMM select for SFENCE from higher privilege modes

ingallsj, 2024-05-07:
The Pointer Masking spec describes

> MPRV and SPVP affect pointer masking as well, causing the pointer masking settings of the effective privilege mode to be applied.

and

> Pointer masking applies to all explicit memory accesses...
> Memory Management: ..., SFENCE.*, HFENCE.*, SINVAL.*, HINVAL.*.

but which PMM should a SFENCE.* instruction select when executed from, say, M-mode?

martinmaas, 2024-05-09:
I think it should be the same behavior as if it was a load or store – i.e., usually the PMM settings for the current privilege mode, or the effective privilege mode as per MPRV and SPVP.

ingallsj, 2024-05-09:
We discussed today that we should add a similar Note for SFENCE.* as what is written for HLV/HSV: https://github.com/riscv/riscv-j-extension/blob/e108fc489cf4f9c5be48e509ce0f96879e782afa/zjpm/instructions.adoc?plain=1#L30-L33

martinmaas, 2024-05-22:
In consultation with the Architecture Review Committee, a decision was reached that these instructions should not apply pointer masking altogether. The rationale is as follows:

Applying pointer masking to SFENCE.* creates a scenario where supervisor code can map an executable code region by modifying the page table, but not apply an SFENCE to it.

Since operating systems are already expected to untag user pointers in software, and since SFENCE often occurs in page fault handlers where the address is already untagged, it was determined that SFENCE.*, HFENCE.*, SINVAL.*, or HINVAL.* should *not* apply pointer masking and code should manually untag addresses before passing them to one of these instructions.

The spec was updated accordingly (654a5c4).

closed 2024-05-22 by martinmaas