riscv/riscv-j-extension #76: sstatus SUM filed still works?

romanheros, 2024-05-09:
The specification doesn't explicitly define under which priviledge pointer masking when status SUM field is 1.  The specification said:
_"Note that the pointer masking setting that is applied only depends on the active privilege mode, not
on the address that is being masked. Some operating systems (e.g., Linux) may use certain bits in
the address to disambiguate between different types of addresses (e.g., kernel and user-mode
addresses). Pointer masking does not take these semantics into account and is purely an arithmetic
operation on the address it is given."_

According to current specification, I think SUM is ingored for pointer masking. Is it right?
Can we clarify which priviledge is using for pointer mask when the SUM  is 1?

jiahzhang, 2024-05-13:
My interpretation is that SUM only affects how the PTE bits are interpreted and not the actual privilege mode of the access, so it would be ignored.

romanheros, 2024-05-14:
> pointer mask when the SUM i

That's also my understanding. Maybe we can take the SUM field as an example on how to understand the sentence I quote above.

jiahzhang, 2024-05-14:
I'm not sure if any update is needed, the spec says that pointer masking only is affected by the effective priv mode. But I'm not writing the spec, so someone else should comment.

martinmaas, 2024-05-22:
We agree that no changes to the spec should be needed. Having said this, thanks for the feedback!

closed 2024-05-22 by martinmaas