riscv/riscv-j-extension #80: MXR for store instructions

BiwefC, 2024-08-15:
> MPRV and SPVP affect pointer masking as well, causing the pointer masking settings of the effective privilege mode to be applied. When MXR is in effect at the effective privilege mode where explicit memory access is performed, pointer masking does not apply.

In pointer-masking v1.0.0-rc2, MXR will effect whether pointer masking is applied or not. But should store instructions is applied with pointer masking be affected by MXR? If so, why should they be affected?

pdonahue-ventana, 2024-08-15:
MXR never affects store instructions.  It makes executable pages readable, not writable.  Pointer masking isn't going to change that.

BiwefC, 2024-08-15:
> MXR never affects store instructions. It makes executable pages readable, not writable. Pointer masking isn't going to change that.

[image]
But in Spike implementation, pmm directly masked by MXR, regardless of the instruction type. Does this means that there is an issue in Spike's implementation?

SiFiveHolland, 2024-08-15:
I think the implication is that yes, MXR affects pointer masking for both both loads and stores, but it doesn't really matter because there's no known use case for doing stores with MXR enabled (execute-only pages aren't writable).

martinmaas, 2024-08-27:
I agree with @SiFiveHolland. I think there is nothing in the spec that says it doesn't apply to stores, but it does not matter.

closed 2024-10-03 by martinmaas