riscv/riscv-j-extension #83: MXR in effect at the effective privilege level

kdockser, 2024-08-28:
After consulting with the Architecture Review Committee, we settled on the following language:

> When MXR is in effect at the effective privilege mode where explicit memory access is performed, pointer masking does not apply.

This clarifies that the MXR field used is determined by the existing RISC-V specification, rather than introducing a special case for pointer masking.

_Originally posted by @martinmaas in https://github.com/riscv/riscv-j-extension/issues/70#issuecomment-2123898322_

kdockser, 2024-08-29:
To further clarify my question, what is the intention of the phrase:
`at the effective privilege mode where explicit memory access is performed`
In
`When MXR is in effect at the effective privilege mode where explicit memory access is performed, pointer masking does not apply.`

Is it implying that the implementation needs to look at the MXR for the effective privilege mode. One of:
- mstatus.MXR
- sstatus.MXR
- vsstatus.MXR

Furthermore, does this MXR disabling of pointer masking extend down to U and VU mode?

martinmaas, 2024-08-30:
Apologies for the slightly delayed response. This was a detail where there was quite a bit of iteration with the Architecture Review Committee. The intention is that whichever MXR field governs a particular memory access (if there is such a field), that MXR field is used to determine whether pointer masking applies (if MXR is set, pointer masking does not apply; if it is not set or there is no governing MXR field, pointer masking does apply).

I believe the two quotes you mentioned in your first comment are consistent and correct in this regard. However, I think the Spike implementation might be slightly incorrect – here is how MXR is determined for regular memory accesses, which I think is the same logic that pointer masking would use as well:

https://github.com/riscv-software-src/riscv-isa-sim/blob/3c5b1bb09ef6daf4146e311d1303cb8dd67c5ff3/riscv/mmu.cc#L407

I think the reason this is correct is that (IIUC) the MXR bit exposed via sstatus is the same as the one in mstatus (the privileged spec states: "The mstatus bit MXR has been exposed to S-mode via sstatus"). I therefore think it is probably correct to only look at sstatus. The case where the current Spike implementation might be incorrect is 2-stage translation, since it is not taken into account here when it should be.

> Is it implying that the implementation needs to look at the MXR for the effective privilege mode.

I believe this is the case. Let me add @aswaterman @Adam-pi3 to also take a look and confirm.

aswaterman, 2024-08-30:
This was fixed in Spike over the last few days.

Adam-pi3, 2024-09-04:
I'm aligned with @martinmaas analysis.

aswaterman, 2024-09-05:
Since the Spike fix is consistent with Martin's analysis, I think we can probably close this one.

closed 2024-09-05 by kdockser

viktoryou, 2025-01-08:
According to the definition of MXR bit in Privilege spec, 
> The MXR (Make eXecutable Readable) bit modifies the privilege with which loads access virtual memory.

I think MXR has no impact when stores access virtual memory. Once MXR is set at the effective privilege mode, whether pointer masking should be applied or not for a store access?