riscv/riscv-j-extension #84: Question on the role of MXR with respect to different transformations

vishmohan, 2024-09-28:
Hi, 

The `Zjpm` spec mentions two different types of transformations:
1. Ignore transformation on the virtual address
2. Ignore transformation on the physical address

It also specifies that `MXR` being set disables pointer masking. Can you clarify if that applies to physical address transformation as well?
Specifically, I am trying to understand the behavior when:
`Mode = M, mstatus.MPRV=0, mstatus.MXR=1 and a load/store from/to physical address is performed.
Does a ignore transformation on physical address  occur in this case?`

```
MPRV and SPVP affect pointer masking as well, causing the pointer masking settings of the effective
privilege mode to be applied. When MXR is in effect at the effective privilege mode where explicit
memory access is performed, pointer masking does not apply.
```
Thank you.

martinmaas, 2024-09-28:
Which of the two operations is performed depends entirely on whether the address is virtual or physical and is not affected by MXR or MPRV. I.e., the fact that it is a physical address has no impact on whether pointer masking is applied, just which of the two operations it is.

In this case, because MXR=1, no pointer masking will be applied.

martinmaas, 2024-10-03:
Although MXR has no effect on *permission checks*, it does have an effect on whether or not pointer masking applies. We agree that these two statements taken together could lead to confusion, and we will thus add the following non-normative statement after the second paragraph quoted above:

> Note that this includes cases where page-based virtual memory is not in effect; i.e., although MXR has no effect on permissions checks when page-based virtual memory is not in effect, it is still used in determining whether or not pointer masking should be applied.

(Please note that this is just adding a non-normative clarifying statement and not a change to the spec.)

closed 2024-12-26 by martinmaas