riscv/riscv-j-extension #86: When PMLEN=16 and hgatp=sv48x4, Is GPA[49:48] =2b00?

chara811, 2024-11-14:


SiFiveHolland, 2024-11-15:
It depends on the VS-stage addressing mode in vsatp.

For vsatp.MODE=Bare, yes. The ignore transformation is applied before any virtual memory lookup. So the GVA has GVA[49:48] = 2b00 and therefore GPA[49:48] = 2b00 as well.

For vsatp.MODE=Sv{39,48,57}, no. The ignore transformation happens before virtual memory, so the virtual address has either GVA[49:48] = 2b00 or GVA[49:48] = 2b11, but the GPA is completely unaffected by pointer masking. It can have any value for those bits.

martinmaas, 2024-12-27:
Thanks @SiFiveHolland for responding to the question. I am closing the issue for now – please re-open if there are any follow-ups.

closed 2024-12-27 by martinmaas