riscv/riscv-j-extension #90: When virtualization is enabled, and the laod instruction accesses the shadow stack page table, does the g-stage require read-write permissions?

chara811, 2024-12-18:


SiFiveHolland, 2024-12-18:
This question sounds like it belongs at https://github.com/riscv/riscv-cfi, where the Shadow Stack (Zicfiss) standard is maintained.

chara811, 2024-12-19:
After reading the spec document, I still have some areas of confusion.
When the shadow stack instruction accesses memory, the g-stage requires read-write permissions. Does the g-stage also need read-write permissions when accessing the shadow stack page with a normal load instruction? This is not clearly stated in the spec. 
Or, for the shadow stack page , is it mandatory for the g-stage to have read-write permissions, and if the g-stage only has read permissions, should it report guest page fault?

martinmaas, 2024-12-27:
I agree with @SiFiveHolland that this is a question about the Zicfiss and not related to pointer masking; pointer masking just transforms the address before any access is performed – the read-write permissions are governed by that access.

Closing this issue for now, but please feel free to re-open if there are any follow-up discussions.

closed 2024-12-27 by martinmaas