riscv/riscv-j-extension #94: How to deal with ignore translation when PMLEN=16 and hgatp.mode equal to Sv48x4

qin996, 2025-03-03:
when we set hgatp.mode equal to Sv48x4, according to RISC-V Privilege spec(version, 20240528), " the size of the incoming address is widened by 2 bits (to 34, 41, 50, or 59 bits)". If vsatp.mode equal to bare mode at the same time, the width of effective address is 50. so in this case, how to deal with ignore translation when PMLEN=16?

SiFiveHolland, 2025-03-04:
This is covered in section 3.5 of the spec:

> Guest physical addresses (GPAs) are 2 bits wider than the corresponding virtual address translation modes, resulting in additional address translation schemes Sv32x4, Sv39x4, Sv48x4 and Sv57x4 for translating guest physical addresses to supervisor physical addresses. When running with virtualization in VS/VU mode with `vsatp.MODE` = Bare, this means that those two bits may be subject to pointer masking, depending on `hgatp.MODE` and `senvcfg.PMM`/`henvcfg.PMM` (for VU/VS mode). If `vsatp.MODE` != BARE, this issue does not apply.

The ignore transformation applies before any virtual memory translation, so bits 49:48 will be zero when doing G-stage translation. The spec suggests multiple options for handling this in hardware:
 * Mask the bits when looking up the entry in the TLB/PTW cache
 * Create TLB/PTW cache entries for all four possible values of those two bits, based on a PTW with the bits being zero.

qin996, 2025-03-05:
_“Smmpm implementations need to satisfy max(largest supported virtual address size, largest supported supervisor physical address size) ⇐ (XLEN - PMLEN) bits to avoid any masking logic on the TLB access path.”_

As described in the spec above, if vsatp = bare and hgatp = sv48x4, VA = GPA. Therefore, the width of VA equal to 50. because GPA has been added 2 bits. If we need to support Smmpm, can PMLEN only be 7, not 16？

thanks

SiFiveHolland, 2025-03-05:
PMLEN can be 16. The spec is saying that, in those conditions, _if_ you support PMLEN=16, there may be a performance impact. (And this performance impact may slow down the processor regardless of the current pointer masking mode.) So you may not want to support PMLEN=16, if you care more about performance than enabling the full set of pointer masking features. But if it makes sense for your particular microarchitecture, you can still support PMLEN=16.

closed 2025-03-07 by qin996