riscv/riscv-j-extension #100: Does mprven in debug mode affect the behavior of pointer masking？

supermessilzb, 2025-04-28:
In debug spec,  when mprven set to 1 in debug mode it describes as follow:

> mprv in mstatus takes effect in Debug Mode

In pointer masking spec, it says：

> MPRV and SPVP affect pointer masking as well, causing the pointer masking settings of the effective privilege mode to be applied.

So mprven in debug mode also affects the behavior of pointer masking for load/store?

pdonahue-ventana, 2025-04-28:
As specified, dcsr.mprven=1 causes MPRV to behave in Debug Mode exactly the same as it behaves in M-mode and dcsr.mprven=0 causes Debug Mode to behave as if MPRV=0.  Anywhere that you would normally apply MPRV, you can instead use `MPRV & (!debug_mode | mprven)`.  This includes pointer masking, endianness, translation, PMP, etc.

supermessilzb, 2025-04-29:
> As specified, dcsr.mprven=1 causes MPRV to behave in Debug Mode exactly the same as it behaves in M-mode and dcsr.mprven=0 causes Debug Mode to behave as if MPRV=0. Anywhere that you would normally [...]

thx!

closed 2025-04-29 by supermessilzb