riscv/riscv-j-extension #101: Which address is used for "ignore" transformation in point masking spec?

supermessilzb, 2025-04-29:
As mentioned in point masking spec:

>  An effective address after the ignore transformation has been applied.

My questions are: 

1. What's the definition of the "effective address"？ For a load/store, is the address in rs1 used as the "effective address", or the address of (rs1+imm)?
2. Is the pointer tag in the upper bits of an address obtained from the address in rs1 or the address of (rs1+imm)?

THX.

martinmaas, 2025-05-05:
Effective address is a term defined in the main RISC-V specification (Section 2.6 of the unprivileged spec), and also in Section 2.1 of the pointer masking spec. It is the address that is used to perform the memory access – i.e., in the example mentioned, it is the address of (rs1+imm).

Regarding the second question: Pointer masking does not define how the tag bits are being used, just that the CPU ignores these bits when performing an access – i.e., there is no notion of obtaining a tag.

supermessilzb, 2025-05-06:
> Effective address is a term defined in the main RISC-V specification (Section 2.6 of the unprivileged spec), and also in Section 2.1 of the pointer masking spec. It is the address that is used to [...]

Thank you, martinmaas.

For my second question, i need to change my desciption in another way: Dose the pointer tag locate in the upper bits of rs1 or the upper bits of (rs1+imm)?

The reason why i raised my second question is that: in my design, if the pointer tag locates in the upper bits of rs1, applying "ignore transformation" to rs1 or to (rs1+imm) would be equivalent; as a result, the implemention of "ignore transformation" would be easier and cheaper if i could choose to applying "ignore transformation" to rs1.

closed 2025-06-18 by martinmaas